mik3y/usb-serial-for-android #634: 16 KB Alignment Warning for Android 15+ Compatibility

softwareintelux, 2025-09-27:
**Summary:**
When building the APK with `usb-serial-for-android:3.9.0`, the following warning is generated:

"APK app-debug.apk is not compatible with 16 KB devices. Some libraries have LOAD segments not aligned at 16 KB boundaries:
lib/arm64-v8a/libbarhopper_v3.so
lib/arm64-v8a/libimage_processing_util_jni.so
Starting November 1st, 2025, all new apps and updates to existing apps submitted to Google Play and targeting Android 15+ devices must support 16 KB page sizes."

This means that the native `.so` files in the library are not aligned to **16 KB boundaries**, which is now a requirement for Android 15+.

**Affected Files:**
- `lib/arm64-v8a/libbarhopper_v3.so`
- `lib/arm64-v8a/libimage_processing_util_jni.so`

**Impact:**
- Internal apps may work now, but Google Play will reject updates for Android 15+ unless this issue is resolved.
- Future Android 15+ device compatibility will break without proper alignment.

**Reference:**
- [Android 16 KB page size requirement](https://developer.android.com/16kb-page-size)

kai-morich, 2025-09-27:
This library does not include native library. The warnings are about libraries from Google.

softwareintelux, 2025-09-29:
Thanks issue resolve